quartiq/miniconf #10: Add setting readback support

ryan-summers, 2021-01-28:
Add support for reading back a staged settings value

jordens, 2021-02-20:
I'm not sure pull readback would be a prime usecase. It would be better to generally ensure that the MQTT view of the active settings is consistent with the device reality. In that sense the solution would be to publish the settings when they are activated. That also addresses:

* Making the active settings available in MQTT when the device boots and there are no retained settings
* Publishing settings changes due to firmware changes
* Validation/implicit rejection of the settings from the staged ones on commit
* Possible interdependencies of settings (the range of one value depending on another).
* Pivoting to the configured settings tree (#25)

ryan-summers, 2021-03-01:
To update, the primary use case here is to publish the default, power-on active settings if no data is received over the MQTT interface so that the active configuration is available via the broker. Settings should be published back to the topic that is used to configure them in the first place.

PR #66 (2021-12-10): Feature/settings republish
This PR closes #10 by adding functionality to the MqttClient to automatically publish all available settings after 2 seconds of connection + last message received.

TODO:
* [x] Refactor static buffer sizes
* [x] Remove iterator state when iteration has completed
* [x] Add MQTT-republish integration test|

Note: https://github.com/quartiq/minimq/issues/74 might mean that a device will fail to publish settings on boot until that issue is fixed and incorporated into a new minimq patch.
closed 2021-12-10 by ryan-summers