quartiq/miniconf #6: Add support for deriving StringSet on enums

ryan-summers, 2021-01-28:
Add support for deriving stringset on custom enums

irwineffect, 2021-02-06:
I've got a few questions about the desired behaviors for enums. Are we only planning on supporting simple enums that do not contain other values?

If we support enums that can contain other values, what would the behavior look like for setting them? For example, let's say we have the following settings structure:

```rust
struct Inner {
    b: u32,
    c: u8,
}

struct Settings {
    a: Option<Inner>,
}
```

If we wanted to set field `b`, would the setting topic be `/a/Some/b`? What should happen if the current type of `a` is `None`, does that throw an error, or force the type of `a` to be `Some`? I'm thinking if the current type is `None`, the user would first need to set `/a` to be `Some(Inner{b: 0, c: 0})` (meaning, fully initialize the nested type), and then they could individually access the nested fields. If we instead want to automatically force to `Some`, I think we would need to require that `Inner` implements the `Default` trait.

ryan-summers, 2021-02-08:
As of right now, serde-json-core doesn't support anything other than simple enums (e.g. C-like enums, no contained values), and I think that's an appropriate approach for us as well.

Assume that any supported enums do not contain any data. We should likely throw a compile error if they do.

PR #17 (2021-02-17): Added support for enums
Closes #6 

Only supports simple enums.
closed 2021-02-17 by ryan-summers