quartiq/miniconf #18: Document leaf node / parent node configuration semantics

ryan-summers, 2021-02-17:
When using Miniconf to set multiple settings at once, this can result in a single value being configured via multiple topics. We need to define which should take priority in the case where multiple `Will` messages are configured for a single setting value.

Say for example, a will message is set for:
`settings/a/b` = 5
and `settings/a` = `{"b" = 6}`

Which should take precedence?

ryan-summers, 2021-02-17:
One question that will be pertinent is to determine which ordering the `Will` messages are received after connection with the MQTT broker as well. If ordering is undefined, the client may need to maintain a state of "will precedence" in determining the precedence level of the previous will configuration.

jordens, 2021-02-17:
I'd rather leave it to the user to implement consistent settings updates.
With precedence you may have to store dirtiness or even old messages as well. E.g. a low precedence larger scope message coming after a high precedence small scope message.

jordens, 2021-02-19:
Alternatively, maybe it's better to specify this in the code: Have an attribute that marks a compound item (I think that only applies to arrays and structs) as being terminal. The default would be to only allow primitive leaf updates.

```rust
struct Settings {
    a0: X,

    #[terminal]
    a1: X,
}

struct X { b: i32 }
```

And the allowed updates would be `settings/a0/b = 1` and `settings/a1 = {"b": 2}`

jordens, 2021-02-22:
Might be beter to place the `terminal` at the `struct X` definition level and not at the `Settings` field level.

PR #27 (2021-03-02): Updating miniconf to reject non-terminal node configurations
This PR fixes #26 by adding a second derive `StringSetAtomic`. This allows for a structure to implement atomic-only configuration. This means that the entire structure must be updated simultaneously. Additionally, there is no `commit` topic/action, as that role is now fulfilled by enforced atomic updates.

Attempting to update a setting when not at the terminus of the settings structure (e.g. an atomic setting, be it a builtin type or a StringSetAtomic-derived type) will now generate an error.

This is done to prevent multiple MQTT t [... cut]
closed 2021-03-02 by ryan-summers