quartiq/miniconf #25: changing root topic at runtime

jordens, 2021-02-20:
Having a runtime configurable root topic is desirable. To be able to pivot from a default root topic based on a unique device ID to a configurable semantic one, the root topic needs to be changed at runtime. This configurable semantic topic would be published on MQTT as well and the device would pivot from the unique default topic to the configured one.

Arises from: https://github.com/quartiq/stabilizer/issues/276

ryan-summers, 2021-03-01:
Current plan is to not yet add support for this, but to potentially add it in the future if needed.

ryan-summers, 2021-04-28:
Closing. MQTT integration has now been removed in favor of offloading these decisions to the application.

closed 2021-04-28 by ryan-summers