quartiq/miniconf #26: Remove the concept of "committing" settings in favor of terminal updates

ryan-summers, 2021-02-22:
Instead of "committing" staged settings into the active settings, @jordens and I came to the conclusion that we likely want to instead:
* Disallow setting the same value from different scopes (https://github.com/quartiq/miniconf/issues/18)
* Inform the application of settings updates whenever a new value arrives

This achieves the same outcome (atomic updates of similar settings) via a simpler mechanism (multiple settings configured simultaneously) and eliminates the strangeness of the `commit` topic.

jordens, 2021-03-01:
Just confirming a design decision: arrays are only atomic through encapsulation in an atomic `struct`.
(correct? @ryan-summers).

ryan-summers, 2021-03-01:
Correct - arrays may not be set entirely at once and will always defer down to their member elements (unless wrapped in MinconfAtomic, as you note)

PR #27 (2021-03-02): Updating miniconf to reject non-terminal node configurations
This PR fixes #26 by adding a second derive `StringSetAtomic`. This allows for a structure to implement atomic-only configuration. This means that the entire structure must be updated simultaneously. Additionally, there is no `commit` topic/action, as that role is now fulfilled by enforced atomic updates.

Attempting to update a setting when not at the terminus of the settings structure (e.g. an atomic setting, be it a builtin type or a StringSetAtomic-derived type) will now generate an error.

This is done to prevent multiple MQTT t [... cut]
closed 2021-03-02 by ryan-summers