quartiq/miniconf #35: Accept minimq messages rather than owning the client

dnadlinger, 2021-03-07:
In many `miniconf` applications, it might be desirable to expose an MQTT interface beyond the auto-generated settings interface (e.g. to trigger sampling of some measurements, query fast-changing status information, etc.).

Currently, `MqttInterface` takes over the `minimq::MqttClient`, so this isn't possible.

The easiest option would probably be to have a `try_process` method instead that the client code can call from the MqttClient message callback, so messages outside the miniconf tree can be handled by application-specific code.

dnadlinger, 2021-03-07:
(This also avoids piping all the network stack/… through MqttInterface, which is a bit clunky.)

dnadlinger, 2021-03-07:
Okay, actually it doesn't remove the need for all the network stack type parameters, as there is no general (non-generic, or only in the error type) MqttInterface trait.

In that instance, we might as well add an extra callback parameter to `update` to which unknown messages are forwarded.

ryan-summers, 2021-04-15:
Just to elaborate - there's also a `miniconf::MqttInterface::client()` function which accepts a closure. That closure is provided the MQTT client object, so you can still perform arbitrary actions directly with the underlying client. This API was intended to allow these types of use-cases that you're referring to.

dnadlinger, 2021-04-15:
But what I need to do is handle messages other than the ones expected by Miniconf. How would that work with the `client()` API?

ryan-summers, 2021-04-15:
That's a valid point - I'll have to think about the architecture here to see if that would make sense

ryan-summers, 2021-04-21:
The more I've thought about this, the more I'm coming under the opinion that owning the client is the wrong API. It makes it really difficult to use MQTT for anything besides miniconf.

I'm going to work on a refactor to pull MQTT ownership out of miniconf. Theoretically, miniconf doesn't even depend on MQTT - it just depends on some topic-data pairing that serializes data with JSON.

PR #39 (2021-04-26): Removing minimq MQTT client ownership
This PR removes MQTT client ownership from `miniconf` in favor of a generic API.

This has the following benefits:
* MQTT can now be used for things other than `miniconf`
* Unsupported messages are no longer silently dropped (fixes #35)
* Any MQTT client (or in fact, any other similar protocol) may be used for miniconf
closed 2021-04-26 by ryan-summers