quartiq/miniconf #51: Constraint validation

jordens, 2021-09-29:
There should be a way to reject semantically invalid values (instead of just syntactically invalid as now).
This is setting-dependent.

* `f32` bounds (max, min)
* Rounding
* Complicated multi-settings constraints (iir filter coefficients leading to undesired zero/pole positions due to rounding)

ryan-summers, 2021-10-05:
We could do this at the derive level using an [attribute macro](https://doc.rust-lang.org/reference/procedural-macros.html#attribute-macros).

I imagine this would look something like the following:

```rust
#[derive(Miniconf)]
struct Settings {
    #[miniconf(validation = "check_voltage")]
    input_voltage: f32,

    output_level: OutputChain,
}

#[derive(MiniconfAtomic)]
#[miniconf(validation = "check_output_chain")]
struct OutputChain {
    pub output_level: f32,
    pub multiplier: u8,
}

fn check_voltage(voltage: f32) -> bool {
   voltage.abs() < 5
}

fn check_output_chain(chain: OutputChain) -> bool {
    if chain.output_level.abs() > 5 {
        return false;
    }

    if chain.multiplier > 4 {
        return false;
    } 

    chain.multiplier as f32 * chain.output_level < 10
}

```

ryan-summers, 2021-10-06:
After further discussion in person, it was brought up that a simpler approach may be to defer acceptance of settings until after the application has a chance to validate them. Control flow would look as follows:

1. Receive and parse settings, store in structure, mark structure as dirty (or store a copy that is not active?)
   * Errors in parsing return an error immediately over MQTT
2. Miniconf informs application of settings update
3. Application schedules the reconfiguration
4. In the future (e.g. different priority task), application attempts to apply settings
5. Application determines if settings application succeeded or failed, informs Miniconf of the result
6. Miniconf then forwards success/fail over MQTT to the configuration request

The main different here is that Miniconf currently sends an acceptance command after step (1), which means that the application cannot reject complex combinations of settings when applying them because Miniconf has already accepted them. This moves the acceptance to depend on a report by the actual application.

ryan-summers, 2021-12-07:
After some thought, it was noted that there's some difficulty in deferring the update to the future. Notably, this means that we would have to cache the following data for after the application of the settings:
1. CorrelationData property
2. Response topic

Instead, the implementation in #63 allows a custom `handler` closure to be passed to `handled_update()` that can be used to validate the settings before accepting them. This means that they can be validated and/or applied immediately after the update and caching of the above data is not required.

PR #63 (2022-02-03): Feature/settings validation
This PR fixes #51 by adding support for a custom handler to be called whenever settings are updated.

Note: Currently, if a new setting is rejected, it still persists in the cached settings.
closed 2022-02-03 by ryan-summers