quartiq/miniconf #54: Using Miniconf with generics doesn't work.

SingularitySurfer, 2021-10-08:
I'm not sure weather this is an Issue with Miniconf but I seem to be unable to use it in a struct with generic parameters.
I'm using MiniConf 0.1.
```
use miniconf::Miniconf;

#[derive(Miniconf)]
pub struct Test<T> {
    pub a: T,
}
```

gives the error:

```
error[E0107]: missing generics for struct `Test`
 --> src/main.rs:4:12
  |
4 | pub struct Test<T> {
  |            ^^^^ expected 1 generic argument
  |
note: struct defined here, with 1 generic parameter: `T`
 --> src/main.rs:4:12
  |
4 | pub struct Test<T> {
  |            ^^^^ -
help: add missing generic argument
  |
4 | pub struct Test<T><T> {
  |            ^^^^^^^

error: aborting due to previous error
```

Is this caused by Miniconf or on my end?

ryan-summers, 2021-10-08:
I don't see how you could derive for a generic argument. There's no way to know the underlying type `T` until it's defined. Theoretically, I guess if you trait-bounded `T: Miniconf + Deserialize` it should _technically_ work, but I don't know if our proc-macro supports this.

Why do you want to use a generic T?

ryan-summers, 2021-10-08:
Similarly, trying to do an unbounded generic derive on `Copy` also fails:
```rust
#[derive(Copy)]
struct Test<T>(T);
```

```
error[E0277]: the trait bound `Test<T>: Clone` is not satisfied
   --> src/main.rs:1:10
    |
1   | #[derive(Copy)]
    |          ^^^^ the trait `Clone` is not implemented for `Test<T>`
    |
    = note: this error originates in the derive macro `Copy` (in Nightly builds, run with -Z macro-backtrace for more info)
```