quartiq/miniconf #71: Settings republish events register as updated settings

ryan-summers, 2021-12-14:
When miniconf republishes settings, it registers the message that it has published as inbound change requests. This causes it to generate settings update events for all settings.

jordens, 2021-12-14:
Yeah. We should probably not just stall incoming messages but actually discard them. Or mark them with a property as application-originating?

ryan-summers, 2021-12-14:
We could theoretically append some `CorrelationData` indicating that they are republished messages.

While we can modify the Miniconf state machine to just reject/ignore any inbound settings configurations while doing republishing, this doens't solve the problem completely since we would transition state after republishing the last setting, which then causes us to re-receive it. I think correlation data is likely the best approach.

PR #82 (2022-02-16): Implementing fix to ignore incoming republish data
This PR fixes #71 by adding correlation data to the republished settings so they can be quickly ignored.

TODO:
- https://github.com/quartiq/minimq/pull/81 needs to be merged and 0.5.2 of minimq needs to be released
closed 2022-02-16 by jordens