quartiq/miniconf #88: Add mutable iteration support

ryan-summers, 2022-05-12:
When iterating across the settings array, the settings structure is immutably borrowed. This means that settings cannot be modified during iteration.

It would be useful to allow mutable iteration of settings.

There are some implications here. If we allow mutably iterating, we will need to specifically address iterating over `Option`s. If the option were replaced during iteration, it could have unintended consequences on structure iteration.

jordens, 2022-05-12:
Do we support `Option`? I though it's only simple `enum`.
Otherwise this problem (mutable walk) would apply to all enums, right?

ryan-summers, 2022-05-12:
We support `Option` as a special case. An `Option` represents a branch of the settings tree that may or may not exist. If the Option is `Some()` then there are settings available down that branch. If it is `None`, that branch is  empty.

We use it for settings that may or may not exist depending on run-time detection. In Booster's case, this is whether or not an RF channel is installed.

jordens, 2022-05-12:
Ah right.
Yeah. On the second point I meant that it would apply to all non-simple enums. But `Option` is the only one supported now.

jordens, 2022-11-02:
I think the only problem here is just that: `Option`.
One approach could be to produce all paths in `next_path` regardless of the runtime state. Then `next_path` could become an associated function. The user (and re-publication) would only have to take care with paths that may not exist at runtime i.e. just try `get()`/`set()` and ask for forgiveness.

ryan-summers, 2022-11-02:
`next_path` takes a `self` parameter, so if the `Option` is `None`, we can't iteratively recurse through the wrapped type.

I think that would be a viable strategy if we removed `self` from `next_path` (I don't think it's required, it's very similar to `metadata()`). Sure there's some annoyance about asking for a value that doesn't actually exist during republication, but that's an easy match-ignore.

jordens, 2022-11-02:
If `next_path` is an associated function, it doesn't take `self`.

PR #105 (2022-11-02): rework (make get/set_path generic Peekable, associated fns for metadata and next_path)
* a small step towards #104
* Changes the path assembly logic to postfix the separator. This is a bit easier in the logic.
* makes `metadata()` an associated fn
* homogenizes the API and e.g. makes the mqtt client and the unittests use `set()` where appropriate
* also makes the metadata semantically correct (depth N if there are N elements in the path)
* make `next_path` and related API associated fns
* Move the iterator state into the iterator
* fixes #88 by iterating all paths regardless of `Option` nulls. [... cut]
closed 2022-11-02 by ryan-summers