quartiq/miniconf #104: be generic over the path format and separator

jordens, 2022-11-01:
No real need to enforce a `Peekable<Split<char>>`.
Maybe https://docs.rs/itertools/latest/itertools/trait.PeekingNext.html is enough.

ryan-summers, 2022-11-01:
There are a few places where a separator is employed:
1. During derivation to construct setting lookup paths
2. During `Miniconf::get()` and `Miniconf::set()` default implementations.
3. In default `miniconf::Array` and `[T; N]` implementations

It seems like the macro could (quite easily) expose a new attribute to specify the separator:
```rust
#[derive(Miniconf)]
#[miniconf(separator=':')]
struct Settings {
   // ...
}
```

During code-gen, we could implement an overriden `get()` and `set()` function as well as updated path generation function. I'm not sure how we could fix up the default implementation for `array` - perhaps the separator could be passed into the `recurse_paths` function.

jordens, 2022-11-01:
Or maybe `next_path` can be restructured to return an iterator over path parts. Then the user (or `iter_paths()`) can join each path.

PR #148 (2023-07-25): be generic over serde mechanism and path hierarchy separator
Close #104 
Close #103 

Tested in `fls` where it also makes the code 1.2 kB smaller.

Downside is that we loose serde error source information.
closed 2023-07-25 by ryan-summers