quartiq/miniconf #106: Investigate wrapper struct usability

ryan-summers, 2022-11-02:
Investigate the usability of `miniconf::{Option, Array}` as opposed to additional `#[miniconf(defer-further)]` or something similar.

Current arguments:
* The current API (Array and Option) allows for easier muxing and nesting of array types
* The attribute-based approach removes the wrapper types, which enables the outer settings struct to implement all kinds of derived traits (e.g. defmt)

ryan-summers, 2022-11-03:
One finding noted by @jordens is that it is not possible `#[derive(Serialize, Deserialize)]` on settings structures that contain the wrapper types

jordens, 2023-07-04:
For `Array` one could consider use `serde_big_array` with additional bounds on `T` (with e.g. `derive_where`).
Having used these in anger and over a good range of applications has been a good experience.
Let's close this and open issues when and if they arise.

closed 2023-07-04 by jordens