quartiq/miniconf #120: `Array` and `Option` have no viable constructors

ryan-summers, 2022-11-10:
The `Array` and `Option` types do not have a mechanism provided to construct them from existing data, but rather only expose `default()` construction.

We need to provide a way to construct these datatypes. This also represents a breaking API change in 0.6.2, which means we should likely yank the release as invalid

jordens, 2022-11-10:
Unfortunately yes, the release was unintentionally breaking. I  don't see a reason to make the field pub. It's not a problem since there are excellent traits that give full access to the inner (deref, deref Mut, into, from, default).

jordens, 2022-11-10:
Summarizing, I think the API is good. But if you think it helps we can yank and bump.

nkrackow, 2022-11-10:
Is this because  [`pub struct Array<T, const N: usize>([T; N]);`](https://github.com/quartiq/miniconf/blob/main/src/array.rs#L20) 
is not `pub struct Array<T, const N: usize>(pub [T; N]);` ?

jordens, 2022-11-10:
Yes

jordens, 2022-11-10:
c.f. https://github.com/quartiq/stabilizer/pull/644 for the breakage and fix

ryan-summers, 2022-12-09:
Closing. We yanked old versions of miniconf to prevent this issue.

closed 2022-12-09 by ryan-summers