quartiq/miniconf #137: Discovery timeout behavior is undesirable

ryan-summers, 2023-04-05:
Currently, we support device discovery. However, it takes a timeout parameter for overall discovery. For devices that may take a long tiem to boot, this timeout doesn't quite behave as we'd like.

In general, discovery with a timeout is only useful with a longer timeout when a device is just being added to the network (i.e. just powered up). In this case, the real desire is to wait up to N seconds for the list of available devices to _change_, and then abort once it does. However, as implemented, the code waits for N seconds even if the devices change after 1 second.

This causes undesirable delays in scripts. We should likely modify the discovery semantics to make it a more usable utility

jordens, 2023-07-04:
The timeout is there to ensure we get all the retained alive messages, not the ones that just come alive while we wait.
And it needs to be longer than the first received since there may be multiple devices.

jordens, 2023-07-04:
But I do think that an adaptive timeout would be better suited and provide a better experience without any downsides: Stop waiting after three times the time it took for the first retained message to be received. Clip the overall timeout to what it is now.

commit 635d74d5 (2024-01-19): py: rework discover

* uses suback RTT to determine timeout for delivery of retained /alive
* due to retained being delivered first,
  an alternative would be to publish non-retained on some topic and wait
  for that to come around, while not specified, this would be exact
* also return a list as the order of retained messages may or may not
  matter

close #137
closed 2024-01-19 by jordens