quartiq/miniconf #142: Interference/problems with miniconf and other mqtt paths

nkrackow, 2023-05-12:
Using the latest master branch with eg. Stabilizer dual-iir (aka using vanilla Stabilizer master branch, just changing the miniconf dep to the github master and minimq to 0.6.2) leads to the following issues:

Miniconf seems to be confused with messages that are not intended for it. I get the following debug prints:
```
INFO - Starting
INFO - EUI48: 80-34-28-1f-59-65
INFO - setup() complete
WARN - Network link DOWN
WARN - Network link UP
INFO - MQTT connected, subscribing to settings
INFO - Unknown Miniconf command: /alive
INFO - Unknown Miniconf command: /telemetry
```

If I publish onto a miniconf setting I get `WARN - Failed to build response message`.

The miniconf settings are applied at the device and I also see the mqtt messages like `/alive` on the host.

jordens, 2023-05-12:
https://github.com/quartiq/miniconf/blob/8be449dcd9be7c838720c21223dac0841595dd26/src/mqtt_client/mod.rs#L324

ryan-summers, 2023-05-12:
I think this may be a knocking effect of the new get/set commands and the telemetry prefixes conflicting and I'm not sure this is a defect in miniconf. Perhaps we need to think more about how the matt tree should look...

nkrackow, 2023-05-22:
Well but we used to always get the response if the setting or the path are invalid which was very helpful. And as it is it always wants to send this but doesn't have a response topic.

jordens, 2023-05-22:
Might be useful for debugging but it's not useful with the miniconf client nor required by the miniconf protocol. Hence it's also not tested or maintained. Let's get rid of it.
The recommendation is to set a response topic (use the miniconf client)

ryan-summers, 2023-06-05:
Agreed with @jordens here. The `ResponseTopic` will provide failure information such as an invalid path or setting. If there's no `ResponseTopic` specified, I don't think we should be doing anything.

PR #146 (2023-06-23): only reply if response topic set
* [x] only try to build response if ResponseTopic is set
* [x] should subscribe to `/settings/#` and `/list` explicitly, not to `/#` (which captures telemetry, alive, and maybe even response topics, log etc as well)

closes #142
closed 2023-06-23 by ryan-summers