quartiq/miniconf #192: Derive `Tree` for structs containing generic `Tree` types

ryan-summers, 2024-03-22:
Currently, it doesn't seem possible to do the following properly:
```rs
struct A<T: TreeKey<Y>, const Y: usize> {
    inner: T,
    outer: u32,
}
```

We would need to annotate the `inner: T` with `#[tree(depth(Y))]`, but the current depth parsing requires an integer. The depth isn't known until compile time, but is known to be Y >= 1. It seems like the depth should optionally take this compile-time generic as well.

ryan-summers, 2024-03-22:
I think a blanket implementation is actually the only way. Arithmetic on const generics (i.e. `impl TreeKey<{Y - 1}>`) is not stable currently.

jordens, 2024-09-30:
One problem would be that a generic field type may not be dominating the overall depth. Then the outer detph would not determine the inner depth in all cases. The offset would have to be constant. And `Y1,Y2` combinations of a struct with multiple inner generic types with generic depth would be impossible unless `Y1 = Y2 + constant`.
A way to do this may be to support `depth: Vec<usize>` on fields and then determine the fixed offset.

closed 2024-10-24 by jordens