quartiq/miniconf #268: Support schema export

airwoodix, 2024-11-29:
Is there tooling or a preliminary work looking into exporting a schema for the settings described by `miniconf`?

In particular for MQTT, something like [asyncapi](https://github.com/asyncapi/spec) could be nice and provide a way to automatically document the settings.

jordens, 2024-11-29:
Not that I'm aware of.

jordens, 2024-11-30:
Easiest approach is probably to co-derive with schemars matching skip, rename etc. and then transform or re-interpret.

airwoodix, 2024-12-02:
Thanks for the feedback! The interaction with `schemars` sounds promising. I guess one could start with being able to derive from `JsonSchema`, and handle the actual schema output separately?

For the `JsonSchema` derivation: one could rely on `schemars`' `serde` compatibilty to write something like:

```rust
#[derive(JsonSchema, Tree)]
struct Settings {
    #[serde(skip)] // or #[schemars(skip)]
    #[tree(skip)]
    foo: Leaf<bool>,
    // ...
}
```

but the duplication is ugly and even more annoying for `rename`.

The easy way around would be to forward the `serde` or `schemars` attributes to the `Tree` derive, but that would heavily break the existing API. Writing the `schemars`/`serde` attributes from the `tree` ones won't work with a derive macro (but with an attribute one). Optionally adding the `JsonSchema` implementation as `Tree` derive is not so easy, because the implementation functions are not exported by `schemars`.

Do you see nicer ways than what's outlined above?

closed 2025-04-20 by jordens